lsongdev/node-escpos #312: support .js file

jqr91, 2020-09-27:
Hi,

does the library support .js file ? I'am writing a .js app but i receive this error:

`  byte = '';
       ^

SyntaxError: Unexpected token =
    at Module._compile (internal/modules/cjs/loader.js:723:23)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
    at Module.load (internal/modules/cjs/loader.js:653:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
    at Function.Module._load (internal/modules/cjs/loader.js:585:3)
    at Module.require (internal/modules/cjs/loader.js:692:17)
    at require (internal/modules/cjs/helpers.js:25:18)
    at Object.<anonymous> (/var/www/html/icomandino-backend/node_modules/escpos/index.js:12:18)
    at Module._compile (internal/modules/cjs/loader.js:778:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
`

is there a solution? 

Thanks!!

gil-obradors, 2020-09-28:
Hi! 

I have the same issue, but I don't have enought knowledges of JS

I'm trying to import this module to Frontend via Webpack...

Do you have any idea?

jor3l, 2020-09-28:
Share more code, I see no reason why it won't work other than the native libraries that are not available in the browser

gil-obradors, 2020-09-29:
HI @jor3l 
I'm trying to transpile with webpack, for the objective of script in frontend. Direct printinc from browser.

I think is for my ignorance, but I  would appreciate it if you could tell me a hint:

`Version: webpack 4.44.2`

```ERROR in ./node_modules/escpos/statuses.js 7:7
Module parse failed: Unexpected token (7:7)
You may need an appropriate loader to handle this file type, currently no loaders are configured to process this file. See https://webpack.js.org/concepts#loaders
| 
| class DeviceStatus {
>   byte = '';
|   bits = [];
|   bitsAsc = [];
 @ ./node_modules/escpos/index.js 12:17-38
 @ ./src/index.js
```

I have @babel/core @babel/preset-env babel-loader installed as devs, but I don't know if must be babel a loader from webpack to transpile this variables into DeviceStatus class...?

Thank you very much!

jor3l, 2020-09-29:
Are you trying to use this with a bluetooth or usb printer connected to the browser?

gil-obradors, 2020-09-29:
In my case not. Network printer.

I'm trying to read about babel and transpiling, to understand what is happening.
I imagine that you use this code for backend with node?

I'm going to share my scenario.

gil-obradors, 2020-09-29:
package.json

``` {
  "name": "testic-escpos",
  "version": "1.0.0",
  "description": "testing",
  "main": "index.js",
  "dependencies": {
    "escpos": "^3.0.0-alpha.6"
  },
  "devDependencies": {
    "@babel/core": "^7.11.6",
    "@babel/preset-env": "^7.11.5",
    "babel-loader": "^8.1.0",
    "webpack": "^4.44.2",
    "webpack-cli": "^3.3.12"
  },
  "scripts": {
    "dev": "webpack --mode development --devtool source-maps",
    "prod": "webpack --mode production"
  },
  "author": "gil",
  "license": "ISC"
}
```

src/index.js
```// import {escpos} from 'escpos'

const escpos = require('escpos');

console.log('hi!')
```

webpack.config.js
```
module.exports = {
    module: {
        rules: [
            {
                test: /\.js$/,
                exclude: /node_modules/,
                use: {loader: 'babel-loader'}
            }
        ]
    }
};
```

.babelrc

```
{
  "presets": [
    ["@babel/preset-env", {
      "targets": {
        "chrome": "85"
      }
    }]
  ]
}
```

Unable to transpile , error () =>
```
(base) gil@gil-NUC10i7FNH:~/WebstormProjects/webpack-base$ npm run dev

> testic-escpos@1.0.0 dev /home/gil/WebstormProjects/webpack-base
> webpack --mode development --devtool source-maps

Hash: 8ea4f49aa479cce9dc3a
Version: webpack 4.44.2
Time: 839ms
Built at: 29/09/2020 21:53:20
      Asset     Size  Chunks                   Chunk Names
    main.js  978 KiB    main  [emitted]        main
main.js.map  862 KiB    main  [emitted] [dev]  main
Entrypoint main = main.js main.js.map
[0] util (ignored) 15 bytes {main} [built]
[1] util (ignored) 15 bytes {main} [built]
[2] stream (ignored) 15 bytes {main} [optional] [built]
[./node_modules/webpack/buildin/global.js] (webpack)/buildin/global.js 472 bytes {main} [built]
[./src/index.js] 87 bytes {main} [built]
    + 100 hidden modules

ERROR in ./node_modules/escpos/statuses.js 7:7
Module parse failed: Unexpected token (7:7)
You may need an appropriate loader to handle this file type, currently no loaders are configured to process this file. See https://webpack.js.org/concepts#loaders
| 
| class DeviceStatus {
>   byte = '';
|   bits = [];
|   bitsAsc = [];
 @ ./node_modules/escpos/index.js 12:17-38
 @ ./src/index.js
npm ERR! code ELIFECYCLE
npm ERR! errno 2
npm ERR! testic-escpos@1.0.0 dev: `webpack --mode development --devtool source-maps`
npm ERR! Exit status 2
npm ERR! 
npm ERR! Failed at the testic-escpos@1.0.0 dev script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/gil/.npm/_logs/2020-09-29T19_53_20_081Z-debug.log

 ```

jor3l, 2020-09-30:
You could extract only what you need from the source and use it without all the other libraries, even the Network adapter is for node, not sure if it will work with a browser. 

Or, try to replace the byte var to something like _byte, I believe the error is because of the word

gil-obradors, 2020-09-30:
Ok! I understand!

NetworkPrinters use net module from Node.
I don't know if there's a transpiler for it.
I have found that epons ePos SDK use socket.io 

I will need to do my own hack for use it in browser...

Thanks for your time!!!! 

PD: The problem about `byte` is not syntaxis, I have changed to _byte and webpack fails again at the same point. Is something related to class constructions I think.   I tried to use browsify to make the bundle js , and then works but fails on require('net') becaouse is a module from node engine.

edwinkcw, 2020-10-07:
I faced this problem when using node 10, after change to use node 12, the error is gone

jqr91, 2020-10-07:
Thanks! It’s works 

Inviato da iPhone

chris-loch, 2020-10-22:
Well, i am using node 12.19.0 and i am getting the exact same "You may need an appropriate loader to handle this file type, currently no loaders are configured to process this file. ..." error. Anybody came up with another solution?

gil-obradors, 2020-10-22:
Not using this.

This is a package for Node engine,
Or try if any compiler that enables  Node modules to Browser JS

Missatge de chris-loch <notifications@github.com> del dia dj., 22 d’oct.
2020 a les 11:59:

closed 2021-03-30 by song940