lsongdev/node-escpos #317: Error: Cannot find module 'escpos-usb'

slexsys, 2020-11-06:
HI.
I run code in documentation and thrown exception

(node:20296) [DEP0005] DeprecationWarning: Buffer() is deprecated due to security and usability issues. Please use the Buffer.alloc(), Buffer.allocUnsafe(), or Buffer.from() methods instead.
(node:20296) UnhandledPromiseRejectionWarning: Error: Cannot find module 'escpos-usb'
Require stack:
- /home/slexsys/projects/esc_pos_ang/electron-quick-start/main.js
- /home/slexsys/projects/esc_pos_ang/electron-quick-start/node_modules/electron/dist/resources/default_app.asar/main.js
- 
    at Module._resolveFilename (internal/modules/cjs/loader.js:961:15)
    at Function.o._resolveFilename (electron/js2c/browser_init.js:257:921)
    at Module._load (internal/modules/cjs/loader.js:844:27)
    at Function.Module._load (electron/js2c/asar.js:769:28)
    at Module.require (internal/modules/cjs/loader.js:1023:19)
    at require (internal/modules/cjs/helpers.js:77:18)
    at escPos (/home/slexsys/projects/esc_pos_ang/electron-quick-start/main.js:24:18)
    at createWindow (/home/slexsys/projects/esc_pos_ang/electron-quick-start/main.js:17:3)
    at /home/slexsys/projects/esc_pos_ang/electron-quick-start/main.js:65:3
(node:20296) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:20296) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.

oliverdotbecker, 2020-11-07:
run:
`npm i escpos-usb`
Then try again.
Also make sure you have read the USB readme: https://github.com/song940/node-escpos/blob/v3/packages/usb/README.md

slexsys, 2020-11-08:
I use electron project.
      git clone https://github.com/electron/electron-quick-start
      cd electron-quick-start 
      npm install
      npm i escpos-usb
      sudo apt-get install build-essential libudev-dev (for Ubuntu 18.04)

Then add code from documentation (https://github.com/song940/node-escpos)
          const escpos = require('escpos');
          escpos.USB = require('escpos-usb');
          const device  = new escpos.USB();
Then
          npm start
And get exception

(node:10171) [DEP0005] DeprecationWarning: Buffer() is deprecated due to security and usability issues. Please use the Buffer.alloc(), Buffer.allocUnsafe(), or Buffer.from() methods instead.
(node:10171) UnhandledPromiseRejectionWarning: Error: Cannot find module 'escpos-usb'
Require stack:
- /home/slexsys/projects/electron-quick-start/main.js
- /home/slexsys/projects/electron-quick-start/node_modules/electron/dist/resources/default_app.asar/main.js
- 
    at Module._resolveFilename (internal/modules/cjs/loader.js:961:15)
    at Function.o._resolveFilename (electron/js2c/browser_init.js:257:921)
    at Module._load (internal/modules/cjs/loader.js:844:27)
    at Function.Module._load (electron/js2c/asar.js:769:28)
    at Module.require (internal/modules/cjs/loader.js:1023:19)
    at require (internal/modules/cjs/helpers.js:77:18)
    at createWindow (/home/slexsys/projects/electron-quick-start/main.js:23:14)
    at /home/slexsys/projects/electron-quick-start/main.js:33:3
(node:10171) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:10171) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.

oliverdotbecker, 2020-11-09:
Did you try to leave the line
`escpos.USB = require('escpos-usb');`
out?

Worked for me as the escpos defines escpos.USB inside the package already and is containing the reffering sources. At least in version 2.4.4, which I'm using.

slexsys, 2020-11-10:
When leave the line
escpos.USB = require('escpos-usb');
out get exception.

(node:6665) [DEP0005] DeprecationWarning: Buffer() is deprecated due to security and usability issues. Please use the Buffer.alloc(), Buffer.allocUnsafe(), or Buffer.from() methods instead.
(node:6665) UnhandledPromiseRejectionWarning: TypeError: escpos.USB is not a constructor
    at createWindow (/home/slexsys/projects/test/electron-quick-start/main.js:22:18)
    at /home/slexsys/projects/test/electron-quick-start/main.js:29:3
(node:6665) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:6665) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.