lsongdev/node-escpos #323: Could not locate the bindings file

Scott1995, 2020-12-08:
Error: Could not locate the bindings file. Tried:
 → D:\demo\project\print\node_modules\_usb@1.6.3@usb\build\usb_bindings.node
 → D:\demo\project\print\node_modules\_usb@1.6.3@usb\build\Debug\usb_bindings.n
ode
 → D:\demo\project\print\node_modules\_usb@1.6.3@usb\build\Release\usb_bindings
.node
 → D:\demo\project\print\node_modules\_usb@1.6.3@usb\out\Debug\usb_bindings.nod
e
 → D:\demo\project\print\node_modules\_usb@1.6.3@usb\Debug\usb_bindings.node
 → D:\demo\project\print\node_modules\_usb@1.6.3@usb\out\Release\usb_bindings.n
ode
 → D:\demo\project\print\node_modules\_usb@1.6.3@usb\Release\usb_bindings.node
 → D:\demo\project\print\node_modules\_usb@1.6.3@usb\build\default\usb_bindings
.node
 → D:\demo\project\print\node_modules\_usb@1.6.3@usb\compiled\14.15.1\win32\x64
\usb_bindings.node
 → D:\demo\project\print\node_modules\_usb@1.6.3@usb\addon-build\release\instal
l-root\usb_bindings.node
 → D:\demo\project\print\node_modules\_usb@1.6.3@usb\addon-build\debug\install-
root\usb_bindings.node
 → D:\demo\project\print\node_modules\_usb@1.6.3@usb\addon-build\default\instal
l-root\usb_bindings.node
 → D:\demo\project\print\node_modules\_usb@1.6.3@usb\lib\binding\node-v83-win32
-x64\usb_bindings.node
    at bindings (D:\demo\project\print\node_modules\?[4m_bindings@1.5.0@bindings
?[24m\bindings.js:126:9)
    at Object.<anonymous> (D:\demo\project\print\node_modules\?[4m_usb@1.6.3@usb
?[24m\usb.js:1:57)
?[90m    at Module._compile (internal/modules/cjs/loader.js:1063:30)?[39m
?[90m    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1092:1
0)?[39m
?[90m    at Module.load (internal/modules/cjs/loader.js:928:32)?[39m
?[90m    at Function.Module._load (internal/modules/cjs/loader.js:769:14)?[39m
?[90m    at Module.require (internal/modules/cjs/loader.js:952:19)?[39m
?[90m    at require (internal/modules/cjs/helpers.js:88:18)?[39m
    at new USB (D:\demo\project\print\node_modules\?[4m_escpos-usb@3.0.0-alpha.4
@escpos-usb?[24m\index.js:28:11)
    at Object.<anonymous> (D:\demo\project\print\download.js:5:17) {
  tries: [
    ?[32m'D:\\demo\\project\\print\\node_modules\\_usb@1.6.3@usb\\build\\usb_bin
dings.node'?[39m,
    ?[32m'D:\\demo\\project\\print\\node_modules\\_usb@1.6.3@usb\\build\\Debug\\
usb_bindings.node'?[39m,
    ?[32m'D:\\demo\\project\\print\\node_modules\\_usb@1.6.3@usb\\build\\Release
\\usb_bindings.node'?[39m,
    ?[32m'D:\\demo\\project\\print\\node_modules\\_usb@1.6.3@usb\\out\\Debug\\us
b_bindings.node'?[39m,
    ?[32m'D:\\demo\\project\\print\\node_modules\\_usb@1.6.3@usb\\Debug\\usb_bin
dings.node'?[39m,
    ?[32m'D:\\demo\\project\\print\\node_modules\\_usb@1.6.3@usb\\out\\Release\\
usb_bindings.node'?[39m,
    ?[32m'D:\\demo\\project\\print\\node_modules\\_usb@1.6.3@usb\\Release\\usb_b
indings.node'?[39m,
    ?[32m'D:\\demo\\project\\print\\node_modules\\_usb@1.6.3@usb\\build\\default
\\usb_bindings.node'?[39m,
    ?[32m'D:\\demo\\project\\print\\node_modules\\_usb@1.6.3@usb\\compiled\\14.1
5.1\\win32\\x64\\usb_bindings.node'?[39m,
    ?[32m'D:\\demo\\project\\print\\node_modules\\_usb@1.6.3@usb\\addon-build\\r
elease\\install-root\\usb_bindings.node'?[39m,
    ?[32m'D:\\demo\\project\\print\\node_modules\\_usb@1.6.3@usb\\addon-build\\d
ebug\\install-root\\usb_bindings.node'?[39m,
    ?[32m'D:\\demo\\project\\print\\node_modules\\_usb@1.6.3@usb\\addon-build\\d
efault\\install-root\\usb_bindings.node'?[39m,
    ?[32m'D:\\demo\\project\\print\\node_modules\\_usb@1.6.3@usb\\lib\\binding\\
node-v83-win32-x64\\usb_bindings.node'?[39m
  ]
}

Scott1995, 2020-12-08:
D:\demo\project\print\node_modules\_bindings@1.5.0@bindings\bindings.js:135
  throw err;
  ^

closed 2020-12-08 by Scott1995

jayan2019, 2020-12-16:
@Scott1995  Have you resolved? If yes how did you resolved?