lsongdev/node-escpos #325: TypeError: Cannot read property of 'indexOf'... trying to connect to SerialPort Fiscal Printer

juanzeta33, 2020-12-16:
Hi guys, 

I dont know if this is an issue of the lib or a missconfig mine, but I'm geting this error when I try to connect to SerialPort printer (currently in emulation) because is a Fiscal Printer (Bussines things in my country).

I'm using COM4 serial port and using this lines as indicates @song940 in this [url ](https://github.com/song940/node-escpos/blob/v3/packages/serialport/README.md)
```
const escpos = require('escpos');
escpos.SerialPort = require('escpos-serialport');

const serialDeviceOnWindows = new escpos.SerialPort('COM10');
const serialDeviceOnLinux = new escpos.SerialPort('/dev/usb/lp0', {
  baudRate: 14400,
  stopBit: 2
});
```

I was happy because is so simple when I connect with networks printer, but I see for SerialPorts printers is another things.

Can you tell is I have something wrong? Tell if I need to provide more code.

My setup is Vue.js with Electron JS to run the printer.
```
"escpos": "^3.0.0-alpha.6",
"escpos-network": "^3.0.0-alpha.5",
"escpos-serialport": "^3.0.0-alpha.4",
```

Image of the error
[image]

closed 2020-12-22 by juanzeta33

Scott1995, 2020-12-23:
How did you solve the problem

juanzeta33, 2020-12-23:
@Scott1995 I'm still working and figthing with them!