lsongdev/node-escpos #330: Error: Port is opening; some other problems

ngvcanh, 2021-01-04:
I using this package with:

- electron: `^11.1.1`
- escpos: `^3.0.0-alpha.6`
- escpos-serialport: `^3.0.0-alpha.4`

My code:

```js
const escpos = require('escpos');
escpos.SerialPort = require('escpos-serialport');

module.exports = {

  printSerial: function(port){
    let device = new escpos.SerialPort('COM2', { baudRate: 19200 });
    let printer = new escpos.Printer(device);

    device.open(function(err){
      console.log('== Err ==', err);
      printer
      .size(1, 1)
      .encode('EUC-KR')
      .align('CT')
      .text('혼밥대왕')
      .align('LT')
      .tableCustom([
        { text: 'Date:', width: 0.4 },
        { text: '2021-01-04 11:11', width: 0.6 }
      ])
      .tableCustom([
        { text: 'Order ID:', width: 0.4 },
        { text: '050-7866-2406', width: 0.6 }
      ])
      .tableCustom([
        { text: '전화번호:', width: 0.4 },
        { text: '200000000', width: 0.6 }
      ])
      .tableCustom([
        { text: '메모:', width: 0.4 },
        { text: '문앞에 두고 벨을 눌러주세요', width: 0.6 }
      ])
      .newLine()
      .text('기사님 이름:')
      .text('기사님 번호:')
      .newLine()
      .tableCustom([
        { text: '내부규', width: 0.33, align: 'LEFT' },
        { text: '내부', width: 0.33, align: 'CENTER' },
        { text: '율과', width: 0.33, align: 'CENTER' }
      ])
      .tableCustom([
        { text: '내부규율과', width: 0.33, align: 'LEFT' },
        { text: '1', width: 0.33, align: 'CENTER' },
        { text: '2,000,000 원', width: 0.33, align: 'CENTER' }
      ])
      .newLine()
      .tableCustom([
        { text: '합계', width: 0.6, align: 'LEFT' },
        { text: '2,000,000 원', width: 0.4, align: 'RIGHT' }
      ])
      .tableCustom([
        { text: '배송료(11.2km)', width: 0.6, align: 'LEFT' },
        { text: '2,000 원', width: 0.4, align: 'RIGHT' }
      ])
      .tableCustom([
        { text: '수수료', width: 0.6, align: 'LEFT' },
        { text: '-0 원', width: 0.4, align: 'RIGHT' }
      ])
      .newLine()
      .tableCustom([
        { text: '총금액', width: 0.6, align: 'LEFT' },
        { text: '2,000,000 원', width: 0.4, align: 'RIGHT' }
      ])
      .newLine()
      .cut();

      setTimeout(function(){
        printer.close();
      }, 1000);
    });
  }

};
```

When I call `printSerial` method, has error

```
== Err == Error: Port is opening
```

But printer still normal print.

Some other problems

- text has black background, 
- It cannot show Korea text.
- How to print dotted line

Thank advance so much for help

DevRafaelCena, 2021-04-05:
I have the same problem .
after printing :  Error: Port is opening

ngvcanh, 2021-04-13:
> I have the same problem .
> after printing : Error: Port is opening

I was re-made this repository at https://github.com/ngvcanh/node-printer