lsongdev/node-escpos #333: Table working incorrect

ngvcanh, 2021-01-09:
I using:

- electron: `^11.1.1`
- escpos: `^3.0.0-alpha.6`

The following code I use to print data for the table

```js
printer
    .tableCustom([
      { text: '메뉴 x 수량', width: 0.3, align: 'LEFT' },
      { text: '금액', width: 0.4,  align: 'RIGHT' }
    ], { encoding: 'EUC-KR' });
    printer.drawLine();
    data.foods.map(food => {
      printer.tableCustom([
        { text: food.food_name + ' x ' +  food.quantity, width: 0.3, align: 'LEFT' },
        { text: food.total_price, width: 0.4, align: 'RIGHT' }
      ], { encoding: 'EUC-KR' });
    });
    printer
    .tableCustom([
      { text: '배달비', width: 0.4, align: 'LEFT' },
      { text: data.delivery_fee, width: 0.4, align: 'RIGHT' }
    ]);
```

And receipt when I print with 2 datas

[image]

[image]

It seems that calculating the text length and cell width doesn't work properly

shraey-chikker, 2022-07-23:
Hey @ngvcanh ,
were u able to find a solution for this?

ngvcanh, 2022-07-24:
> Hey @ngvcanh , were u able to find a solution for this?

I haven't worked with it for a long time. However I did rewrite some of the code, check it out

https://github.com/ngvcanh/node-printer