lsongdev/node-escpos #334: Can't print special characters even with correct encoding

angelod1as, 2021-01-22:
My code is:

```ts
import escpos from 'escpos'
import USB from 'escpos-usb'

export default function nodeEscpos(): void {
  const device = new USB()

  const options = { encoding: 'iso-8859-1' /* default */ }

  const printer = new escpos.Printer(device, options)

  device.open(function (error: any) {
    if (error) {
      console.log(error)
      return
    }
    printer
      .text(
        'À noite, vovô Kowalsky vê o ímã cair no pé do pinguim queixoso e vovó põe açúcar no chá de tâmaras do jabuti feliz',
      )
      .cut()
      .close()
  })
}
```

I'm trying to print this text, that has all letters of the alphabet, including special ones in Portuguese. I tried A LOT of different encodings — iso-8859-1 or UTF-8 should've worked right away — but all I get is weird characters.

I'm using a generic POS-5890C printer and:

```json
    "escpos": "^3.0.0-alpha.6",
    "escpos-usb": "^3.0.0-alpha.4",
```

andrefmandrade, 2021-03-23:
I'm having the same problem, tried all types of encoding and none of them displayed special Portuguese characters.

angelod1as, 2021-03-23:
@andrefmandrade I ended up finding a solution to this. I just can't remember exactly which.

Anyways, see my repo with working code as reference. https://github.com/angelod1as/poem-printer

closed 2021-03-23 by angelod1as