lsongdev/node-escpos #335: Linux common workflow

oliverdotbecker, 2021-01-23:
Hello,

I'm using a HOIN HOP-E802 Printer. For linux they are providing no USB driver. Only a driver working with CUPS. Is there a way to send the commands over CUPS to such a generic printer, or does somebody have a source to a genreric  80mm USB printer dirver fpr linux.

I have it working for Windows. But can't get it going on a Raspberry PI 3 (raspbian).

Ideas?

oliverdotbecker, 2021-01-25:
Never mind.

closed 2021-01-25 by oliverdotbecker

ihsankl, 2022-02-08:
did you make it? mine isn't working at all

oliverdotbecker, 2022-02-08:
Here is mý documentation how I got it work. It's a year ago that I've wrote this, and currently I'm out of time to double check. Hope this helps you
[How to install driver.md](https://github.com/song940/node-escpos/files/8027857/How.to.install.driver.md)
.