lsongdev/node-escpos #336: v3 and WebUSB

yelhouti, 2021-01-28:
First of all than you for the great library and for the work done on moving `require` to the part where it's needed.
This means that I can now, use the library inside a web site using the WebUSB API.

What I'm trying to do now, is write a WebUSB adapter, that takes as a parameter a [USBDevice](https://developer.mozilla.org/en-US/docs/Web/API/USBDevice), and the endpoint to use to [transferOut](https://developer.mozilla.org/en-US/docs/Web/API/USBDevice/transferOut) the content of the buffer to the device.

I would really appreciate some help working on this adapter, from any one with a working device that uses the vanilla USB adapter, since I don't have a thermal printer at my disposal for quick coding/testing/debugging.

georgegrigorita, 2021-03-02:
I have a termal printer and I can help. I'm starting to work on moving away from qz-tray that acts like a printing server to WebUSB and I was looking for a way to integrate escpos with webusb.

yelhouti, 2021-03-02:
I have it working here: https://github.com/yelhouti/escpos-webusb :D

yelhouti, 2021-03-02:
I will be happy to have my code merged here, in the meanwhile please do not squat the npm registry and leave it for the project

globaltechnology-id, 2021-04-01:
@yelhouti how about this library? Do you have the amd version? I mean i can only print text using webusb, i want styling like image, barcode etc without learning the command in hexa or byte format

yelhouti, 2021-04-01:
what do you mean by how about "THIS" library ? my lib is an "extention" of THIS one

globaltechnology-id, 2021-04-01:
I dont know much about javascript, i mean the amd?commonjs? Or whatever it called, so i can add the js file in my laravel project and include it in the blade file.

sibelius, 2022-08-19:
can you send this as a package inside this repo?

ferminc, 2022-08-29:
Hey, pretty late to the party but anyone who wants a working package with WebUSB support should consider using 'escpos-buffer' https://github.com/grandchef/escpos-buffer

sibelius, 2022-08-29:
is this package the same as this one? or another implementation?

ferminc, 2022-08-29:
> is this package the same as this one? or another implementation?

It works independently