lsongdev/node-escpos #337: Please include an option to change the width of the paper

mohamad-supangat, 2021-02-05:
I use this module and get a print that is too long on each line, so it's not neat
[image]

and my printer should use a size of 30 instead of 48
this is the result
[image]

I manually edit in node_modules
[image]
[image]
[image]

Please include an option to set the character size per line along with the encoding options

eclyptox, 2021-02-05:
As you can see here. If you pass in printer options the width. Yo do not need to change library.
[image]

```
const options = {
    encoding: "GB18030",
    width: 30
}
const printer = new escpos.Printer(device, options);
```

mohamad-supangat, 2021-02-09:
thanks , apparently I forgot to check the version that I am using, and I am using the old version ^ 2.5.2 in package.json