lsongdev/node-escpos #339: How can i print UTF-8 characters ?

teenking, 2021-02-18:
Dear All,
How can i print UTF-8 characters ?
It's print only question mark ???????
Please Help....

SpaceDux, 2021-03-04:
Try;

var options = { encoding: "UTF8"};

Options for you Device Opening;

Example;
printer = new escpos.Printer(device, options);

andrefmandrade, 2021-03-23:
> Try;
> 
> var options = { encoding: "UTF8"};
> 
> Options for you Device Opening;
> 
> Example;
> printer = new escpos.Printer(device, options);

UTF8 is not working somehow ...

andrefmandrade, 2021-03-23:
> > > Try;
> > > var options = { encoding: "UTF8"};
> > > Options for you Device Opening;
> > > Example;
> > > printer = new escpos.Printer(device, options);
> > 
> > 
> > UTF8 is not working somehow ...
> 
> It's working for me, can you show me your code?

`const escpos = require("escpos");

escpos.Network = require("escpos-network");

const device = new escpos.Network("192.168.100.55");

const options = { encoding: "utf8" };

const printer = new escpos.Printer(device, options);

device.open(function (err) {
  printer.align("ct");
  printer.text(
    "À noite, vovô Kowalsky vê o ímã cair no pé do pinguim queixoso e vovó põe açúcar no chá de tâmaras do jabuti feliz"
  );
  printer.drawLine("-");
  printer.feed();
  printer.feed();
  printer.feed();
  printer.cut().close();
});
`

SpaceDux, 2021-03-23:
> > > > Try;
> > > > var options = { encoding: "UTF8"};
> > > > Options for you Device Opening;
> > > > Example;
> > > > printer = new escpos.Printer(device, options);
> > > 
> > > 
> > > UTF8 is not working somehow ...
> > 
> > 
> > It's working for me, can you show me your code?
> 
> ```
> 
> escpos.Network = require("escpos-network");
> 
> const device = new escpos.Network("192.168.100.55");
> 
> const options = { encoding: "utf8" };
> 
> const printer = new escpos.Printer(device, options);
> 
> device.open(function (err) {
>   printer.align("ct");
>   printer.text(
>     "À noite, vovô Kowalsky vê o ímã cair no pé do pinguim queixoso e vovó põe açúcar no chá de tâmaras do jabuti feliz"
>   );
>   printer.drawLine("-");
>   printer.feed();
>   printer.feed();
>   printer.feed();
>   printer.cut().close();
> });
> ```
> 
> This is what i get
> https://i.imgur.com/iXCHvwx.jpg

Try this, let me know if it works.

`escpos.Network = require("escpos-network");

const device = new escpos.Network("192.168.100.55");

const options = { encoding: "utf8" };

const printer = new escpos.Printer(device, options);

device.open(function (err) {
  printer.align("ct");
  printer.encode("utf8");
  printer.text(
    "À noite, vovô Kowalsky vê o ímã cair no pé do pinguim queixoso e vovó põe açúcar no chá de tâmaras do jabuti feliz"
  );
  printer.drawLine("-");
  printer.feed();
  printer.feed();
  printer.feed();
  printer.cut().close();
});`

SpaceDux, 2021-03-23:
[image]

Ignore the poor quality, my printer is in need of replacing.
Replicated, and works my end I believe?

andrefmandrade, 2021-03-23:
Text don't match with the printed result.
We should see some accentuation on the printed version also.

andrefmandrade, 2021-03-23:
> Try this, let me know if it works.
>
> ```
> escpos.Network = require("escpos-network");
>
> const device = new escpos.Network("192.168.100.55");
>
> const options = { encoding: "utf8" };
>
> const printer = new [...]

The code sample you sent didn't work aswell

ehababdalaalelhaj, 2021-07-13:
I have the same problem, any luck solving it ?

andrefmandrade, 2021-07-14:
> I have the same problem, any luck solving it ?

unfortunately no, I had to remove every accentuation

andrefmandrade, 2021-07-14:
> I have the same problem, any luck solving it ?

I just figured it out.
Set the encoding to 860 instead of utf8.

`const options = { encoding: '860' };`

GuilhermeEK23, 2025-01-10:
> > I have the same problem, any luck solving it ?
> 
> I just figured it out. Set the encoding to 860 instead of utf8.
> 
> `const options = { encoding: '860' };`

I used 860, but it was supposed to come out áóéíõãçà and it came out áóéíöäçà