lsongdev/node-escpos #341: libusb_Error_Access linux while opning electron app

sidhdharth-navsoft, 2021-02-23:
getting this error while opening electron app by clicking , when we are opening electron app through terminal using sudo app works perfectly, we also tried sudo prompt but it is  not working geeting same error libusb_Error_Access.

const sudo = require('sudo-prompt');
app.on("ready", () => {
    const options = {
        name: 'Electron'
    };
    sudo.exec('apt-get install libudev-dev', options,
        (error, stdout, stderr) => {
            if (error) throw error;
            log.info('-->cmd sudo app running.');
            log.info('stderr: ' + stderr);
            log.info('stdout: ' + stdout);
        }
    );
});

closed 2021-03-03 by sidhdharth-navsoft

sidhdharth-navsoft, 2021-03-03:
i have fix this by allowing all usb port in linux usb rule from terminal

zakaria-chahboun, 2021-07-18:
### Linux Print without *sudo permission*
Add a new file in:
- `/etc/udev/rules.d` named `usb.rules`

It should have the following contents:
- `SUBSYSTEM=="usb", MODE="0666"`

You can do this in terminal quickly: `sudo nano /etc/udev/rules.d/usb.rules`

Reload rules:
- `sudo udevadm control --reload`

> This will make all USB devices readable and writable by all users.