lsongdev/node-escpos #345: Multiple sizes

rodolphocarvalho, 2021-03-29:
Hi, I can't use multiple sizes in the same print when I change one, every one is chaging sizes

`  .style('B')
  .align('ct')
  .size(2,1)
  .text('Example text 1')
  .size(1,1)
  .text('Example text 2')
  .feed()`

The two texts are printed in the size(2,1

how can I solve that?

andrefmandrade, 2021-03-29:
> Hi, defining mutiple sizes is not working, how can I set different sizes for two different texts? I tried like that but it prints all in size(2,1)
> 
> ```
>  .style('B')
>   .align('ct')
>   .size(2,1)
>   .text('Example text 1')
>   .size(1,1)
>   .text('Example text 2')
>   .feed()
> ```
> 
> how can I solve that?

It should work default like this ...

```
  printer.align("CT");
  printer.feed();
  printer.size(0, 0);
  printer.text("SIZE 0");
  printer.size(1, 1);
  printer.text("SIZE 1");
  printer.size(2, 2);
  printer.text("SIZE 2");
  printer.size(3, 3);
  printer.text("SIZE 3");
  printer.size(0, 0);
  printer.drawLine();
  printer.size(2, 1);
  printer.text("SIZE 2,1");
  printer.size(0, 1);
  printer.text("SIZE 0,1");
  printer.size(3, 4);
  printer.text("SIZE 3, 4");
  printer.size(5, 6);
  printer.text("SIZE 5,6");
```

[Output](https://i.imgur.com/Eu6tCs3.png)

andrefmandrade, 2021-03-29:
Also, the printer.size() works like a global configuration. _(At least for me and my printer)_
Lets say you are printing something like shown on the example, and you want your next print to be size(0,0) you need to set it again in the beginning, or it will have the last size configuration.

closed 2021-03-29 by rodolphocarvalho

rodolphocarvalho, 2021-03-29:
That is the point, I was trying to reset the config inside of the first print scope, thanks