lsongdev/node-escpos #346: Images with whit straps

rodolphocarvalho, 2021-03-29:
Hi, when I try to print images, they are getting white stripes I tried other images and the same issue happens the code is that what could be happening?

```
const logo = path.join(__dirname, 'tux.png');
escpos.Image.load(logo, function(image){

  networkDevice.open(function(){

    printer
    .align('ct')
           .image(image, 's8')
           .then(() => { 
              printer.cut().close(); 
           });

    // OR non-async .raster(image, "mode") : printer.text("text").raster(image).cut().close();

  });

});

![20210329_155008](https://user-images.githubusercontent.com/66786177/112884940-7e211400-90a6-11eb-98e8-b74cf830de97.jpg)

```

rodolphocarvalho, 2021-03-30:
I solved the problem using that

`printer.raster(image)`

and not

```
.image(image, 's8')
   .then(() => { 
    printer.cut().close(); 
});
```

closed 2021-03-30 by rodolphocarvalho

goclases03, 2024-10-23:
> Hola, cuando intento imprimir imágenes, aparecen rayas blancas. Probé con otras imágenes y ocurre el mismo problema. ¿Cuál es el código que podría estar pasando?
> 
> ```
> const logo = path.join(__dirname, 'tux.png');
> escpos.Image.load(logo, function(image){
> 
>   networkDevice.open(function(){
> 
>     printer
>     .align('ct')
>            .image(image, 's8')
>            .then(() => { 
>               printer.cut().close(); 
>            });
> 
>     // OR non-async .raster(image, "mode") : printer.text("text").raster(image).cut().close();
> 
>   });
> 
> });
> ```
> 
> https://user-images.githubusercontent.com/66786177/112884940-7e211400-90a6-11eb-98e8-b74cf830de97.jpg

use printer.raster(image);