lsongdev/node-escpos #348: How to use/find printer by name?

challarao, 2021-03-31:
Is it possible give printer name. 

Use case:
You already added printer to the system(usb, network etc) and named it at system level. I see this option in electron pos printer.

Is the serial option the same?

zakaria-chahboun, 2021-07-22:
i guess you can retrieve the printer info by using electron [printer info](https://www.electronjs.org/docs/api/structures/printer-info)

rodchen-king, 2022-01-08:
Is there any solution?

I want to provide the options by printer for customer to select. and then print content.

rodchen-king, 2022-01-17:
> Is it possible give printer name.
> 
> Use case: You already added printer to the system(usb, network etc) and named it at system level. I see this option in electron pos printer.
> 
> Is the serial option the same?

Maybe you can use printer npm. I have use it.