lsongdev/node-escpos #356: escpos.Serial is not a constructor

rareh21, 2021-07-15:
Hi,
I am trying to connect using serial, but it shows escpos.Serial is not a constructor. How to fix this issue. Please suggest me a way to solve this. Thanks

pilz97, 2021-07-26:
I have the same issue

ehmad, 2021-08-03:
I have the same issue

johnstan, 2022-01-05:
I have the same issue

song940, 2022-01-05:
https://github.com/song940/node-escpos/blob/v3/packages/serialport/README.md

johnstan, 2022-01-05:
Ah nice, thanks! Assume after changing the serialPort part, escpos.Screen etc. goes as per the normal docs?

johnstan, 2022-01-05:
Hmm, no - now escpos.Screen is not a constructor. This used to work fine, but now all of the screen commands in the examples seem to be broken

johnstan, 2022-01-06:
Ok, got it - similar trick for the screen as for the serial. 

```npm i escpos-screen``` and then
```escpos.Screen = require('escpos-screen')```

Leaving this here for anyone who searches in the future

closed 2022-01-06 by song940

ma-zal, 2022-04-10:
If you are using `escpos` version 2, there is an issue with new major release of library `serialport` (version 10 with breaking changes has been released).

See my https://github.com/song940/node-escpos/issues/383#issuecomment-1094356051