lsongdev/node-escpos #364: not able to install escpos-usb drivers getting gyp error

HuNtErX6969, 2021-09-06:
npm ERR! code 1
npm ERR! path E:\work\chetna node\chetnabackend\node_modules\usb
npm ERR! command failed
npm ERR! command C:\Windows\system32\cmd.exe /d /s /c prebuild-install --runtime napi --target 4 --verbose || node-gyp rebuild
npm ERR! prebuild-install info begin Prebuild-install version 5.3.6
npm ERR! prebuild-install info install installing standalone, skipping download.
npm ERR! gyp info it worked if it ends with ok
npm ERR! gyp info using node-gyp@7.1.2
npm ERR! gyp info using node@14.17.6 | win32 | x64
npm ERR! gyp info find Python using Python version 2.7.15 found at "C:\Python27\python.exe"
npm ERR! gyp ERR! find VS 
npm ERR! gyp ERR! find VS msvs_version was set from command line or npm config
npm ERR! gyp ERR! find VS - looking for Visual Studio version 2015
npm ERR! gyp ERR! find VS VCINSTALLDIR not set, not running in VS Command Prompt
npm ERR! gyp ERR! find VS checking VS2019 (16.11.31624.102) found at:
npm ERR! gyp ERR! find VS "C:\Program Files (x86)\Microsoft Visual Studio\2019\Community"
npm ERR! gyp ERR! find VS - found "Visual Studio C++ core features"
npm ERR! gyp ERR! find VS - found VC++ toolset: v142
npm ERR! gyp ERR! find VS - found Windows SDK: 10.0.19041.0
npm ERR! gyp ERR! find VS - msvs_version does not match this version
npm ERR! gyp ERR! find VS could not find a version of Visual Studio 2017 or newer to use
npm ERR! gyp ERR! find VS looking for Visual Studio 2015
npm ERR! gyp ERR! find VS - not found
npm ERR! gyp ERR! find VS not looking for VS2013 as it is only supported up to Node.js 8
npm ERR! gyp ERR! find VS
npm ERR! gyp ERR! find VS valid versions for msvs_version:
npm ERR! gyp ERR! find VS - "2019"
npm ERR! gyp ERR! find VS - "C:\Program Files (x86)\Microsoft Visual Studio\2019\Community"
npm ERR! gyp ERR! find VS
npm ERR! gyp ERR! find VS **************************************************************
npm ERR! gyp ERR! find VS You need to install the latest version of Visual Studio
npm ERR! gyp ERR! find VS including the "Desktop development with C++" workload.
npm ERR! gyp ERR! find VS For more information consult the documentation at:
npm ERR! gyp ERR! find VS https://github.com/nodejs/node-gyp#on-windows
npm ERR! gyp ERR! find VS **************************************************************
npm ERR! gyp ERR! find VS
npm ERR! gyp ERR! configure error
npm ERR! gyp ERR! stack Error: Could not find any Visual Studio installation to use
npm ERR! gyp ERR! stack     at VisualStudioFinder.fail (C:\Users\mayur\AppData\Roaming\npm\node_modules\npm\node_modules\node-gyp\lib\find-visualstudio.js:121:47)
npm ERR! gyp ERR! stack     at C:\Users\mayur\AppData\Roaming\npm\node_modules\npm\node_modules\node-gyp\lib\find-visualstudio.js:74:16
npm ERR! gyp ERR! stack     at VisualStudioFinder.findVisualStudio2013 (C:\Users\mayur\AppData\Roaming\npm\node_modules\npm\node_modules\node-gyp\lib\find-visualstudio.js:351:14)
npm ERR! gyp ERR! stack     at C:\Users\mayur\AppData\Roaming\npm\node_modules\npm\node_modules\node-gyp\lib\find-visualstudio.js:70:14
npm ERR! gyp ERR! stack     at C:\Users\mayur\AppData\Roaming\npm\node_modules\npm\node_modules\node-gyp\lib\find-visualstudio.js:372:16
npm ERR! gyp ERR! stack     at C:\Users\mayur\AppData\Roaming\npm\node_modules\npm\node_modules\node-gyp\lib\util.js:54:7
npm ERR! gyp ERR! stack     at C:\Users\mayur\AppData\Roaming\npm\node_modules\npm\node_modules\node-gyp\lib\util.js:33:16
npm ERR! gyp ERR! stack     at ChildProcess.exithandler (child_process.js:397:5)
npm ERR! gyp ERR! stack     at ChildProcess.emit (events.js:400:28)
npm ERR! gyp ERR! stack     at maybeClose (internal/child_process.js:1055:16)
npm ERR! gyp ERR! System Windows_NT 10.0.19043
npm ERR! gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Users\\mayur\\AppData\\Roaming\\npm\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
npm ERR! gyp ERR! cwd E:\work\chetna node\chetnabackend\node_modules\usb
npm ERR! gyp ERR! node -v v14.17.6
npm ERR! gyp ERR! node-gyp -v v7.1.2
npm ERR! gyp ERR! not ok

//note
i have install visual studio 2019 community edition and have selected all the desktop c++ workload and universal tools too.... tried every other solution

Ans1998, 2021-09-13:
使用 `yarn add escpos` 把包升级到  `"escpos": "^3.0.0-alpha.6"`，同时你要确认你的Microsoft Visual Studio环境是没有问题的。
[image]

如果你配合了` printer`包，你需要把` printer`移除`package.json`，等全部依赖安装好了在执行`npm install printer --legacy-peer-deps`
编译的时候可能会提示
[image]
把它删掉即可
[image]

我的:
node:v16.9.0
node-gyp:v8.2.0

RFebrians, 2023-09-18:
same , i hit this when i tried to install 
node : v16.17.1
node-gyp : v9.4.0