lsongdev/node-escpos #365: yarn bluetooth error

lulian, 2021-09-24:
Output:
gyp info it worked if it ends with ok
gyp info using node-gyp@5.1.0
gyp info using node@12.22.1 | darwin | x64
gyp info find Python using Python version 2.7.16 found at "/System/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python"
gyp WARN EACCES current user ("lulian") does not have permission to access the dev dir "/Users/lulian/Library/Caches/node-gyp/12.22.1"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/var/folders/b9/ff3n4b213xs5bf1vkcx38fk80000gn/T/.node-gyp"
gyp info spawn /System/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python
gyp info spawn args [
gyp info spawn args   '/usr/local/lib/node_modules/npm/node_modules/node-gyp/gyp/gyp_main.py',
gyp info spawn args   'binding.gyp',
gyp info spawn args   '-f',
gyp info spawn args   'make',
gyp info spawn args   '-I',
gyp info spawn args   '/Users/lulian/Downloads/node-escpos-3/packages/bluetooth/node_modules/node-bluetooth/build/config.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/usr/local/lib/node_modules/npm/node_modules/node-gyp/addon.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/var/folders/b9/ff3n4b213xs5bf1vkcx38fk80000gn/T/.node-gyp/12.22.1/include/node/common.gypi',
gyp info spawn args   '-Dlibrary=shared_library',
gyp info spawn args   '-Dvisibility=default',
gyp info spawn args   '-Dnode_root_dir=/var/folders/b9/ff3n4b213xs5bf1vkcx38fk80000gn/T/.node-gyp/12.22.1',
gyp info spawn args   '-Dnode_gyp_dir=/usr/local/lib/node_modules/npm/node_modules/node-gyp',
gyp info spawn args   '-Dnode_lib_file=/var/folders/b9/ff3n4b213xs5bf1vkcx38fk80000gn/T/.node-gyp/12.22.1/<(target_arch)/node.lib',
gyp info spawn args   '-Dmodule_root_dir=/Users/lulian/Downloads/node-escpos-3/packages/bluetooth/node_modules/node-bluetooth',
gyp info spawn args   '-Dnode_engine=v8',
gyp info spawn args   '--depth=.',
gyp info spawn args   '--no-parallel',
gyp info spawn args   '--generator-output',
gyp info spawn args   'build',
gyp info spawn args   '-Goutput_dir=.'
gyp info spawn args ]
gyp info spawn make
gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
  CXX(target) Release/obj.target/BluetoothSerialPort/src/osx/DeviceINQ.o
../src/osx/DeviceINQ.mm:75:16: warning: 'Call' is deprecated [-Wdeprecated-declarations]
        found->Call(2, argv);
               ^
../../nan/nan.h:1743:3: note: 'Call' has been explicitly marked deprecated here
  NAN_DEPRECATED inline v8::Local<v8::Value>
  ^
../../nan/nan.h:108:40: note: expanded from macro 'NAN_DEPRECATED'
# define NAN_DEPRECATED __attribute__((deprecated))
                                       ^
../src/osx/DeviceINQ.mm:78:15: warning: 'Call' is deprecated [-Wdeprecated-declarations]
    callback->Call(0, 0);
              ^
../../nan/nan.h:1743:3: note: 'Call' has been explicitly marked deprecated here
  NAN_DEPRECATED inline v8::Local<v8::Value>
  ^
../../nan/nan.h:108:40: note: expanded from macro 'NAN_DEPRECATED'
# define NAN_DEPRECATED __attribute__((deprecated))
                                       ^
../src/osx/DeviceINQ.mm:109:16: warning: 'Call' is deprecated [-Wdeprecated-declarations]
    baton->cb->Call(1, argv);
               ^
../../nan/nan.h:1743:3: note: 'Call' has been explicitly marked deprecated here
  NAN_DEPRECATED inline v8::Local<v8::Value>
  ^
../../nan/nan.h:108:40: note: expanded from macro 'NAN_DEPRECATED'
# define NAN_DEPRECATED __attribute__((deprecated))
                                       ^
../src/osx/DeviceINQ.mm:121:28: error: expected ')'
void DeviceINQ::Init(Handle<Object> target) {
                           ^
../src/osx/DeviceINQ.mm:121:21: note: to match this '('
void DeviceINQ::Init(Handle<Object> target) {
                    ^
../src/osx/DeviceINQ.mm:121:17: error: out-of-line definition of 'Init' does not match any declaration in 'DeviceINQ'
void DeviceINQ::Init(Handle<Object> target) {
[... 132 lines cut ...]
  ^
/var/folders/b9/ff3n4b213xs5bf1vkcx38fk80000gn/T/.node-gyp/12.22.1/include/node/v8config.h:328:29: note: expanded from macro 'V8_DEPRECATED'
  declarator __attribute__((deprecated(message)))
                            ^
../src/osx/DeviceINQ.mm:325:9: error: too few arguments to function call, expected 4, have 3; did you mean 'Nan::Call'?
    cb->Call(Nan::GetCurrentContext()->Global(), 1, argv);
        ^~~~
        Nan::Call
../../nan/nan_maybe_43_inl.h:345:30: note: 'Nan::Call' declared here
inline MaybeLocal<v8::Value> Call(
                             ^
13 warnings and 12 errors generated.
make: *** [Release/obj.target/BluetoothSerialPort/src/osx/DeviceINQ.o] Error 1
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (events.js:314:20)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:276:12)
gyp ERR! System Darwin 20.3.0
gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "configure" "build"
gyp ERR! cwd /Users/lulian/Downloads/node-escpos-3/packages/bluetooth/node_modules/node-bluetooth
gyp ERR! node -v v12.22.1
gyp ERR! node-gyp -v v5.1.0
gyp ERR! not ok
info Visit https://yarnpkg.com/en/docs/cli/install for documentation about this command.

ihsankl, 2022-02-07:
using node version 10 seems to be able to install it. but i can make it work.

jeremy-hoot, 2022-02-23:
I am getting the same error. Were you able to move past this or find a fix?