lsongdev/node-escpos #368: Trying to use with React JS

dsall, 2021-11-03:
I have a sass webapp for pos solution. I am trying to use this library to allow my webapp to print automatically to the network thermal printer. 
when i require the escpos package it shows me a bunch of errors. 
Has everyone used this library with React or know how i can go about it.

dsall, 2021-11-03:
This is the error it shows me.

./node_modules/escpos/statuses.js 6:7
Module parse failed: Unexpected token (6:7)
File was processed with these loaders:
 * ./node_modules/babel-loader/lib/index.js
You may need an additional loader to handle the result of these loaders.
|
| class DeviceStatus {
>   byte = '';
|   bits = [];
|   bitsAsc = [];

elliehachemjob, 2021-11-03:
usually babel the default one you get with react app will not be able to convert your Sass code so you need to install a loader that can convert Sass synatx  https://webpack.js.org/loaders/sass-loader/

dsall, 2021-11-03:
I probably misquoted.  I am not sure if that's the issue. But  i will look into what you said.

basherahmadahsas, 2021-11-09:
@dsall I am also facing the same error, did you find any solution if please share it with me as well.

dsall, 2021-11-09:
> @dsall I am also facing the same error, did you find any solution if please share it with me as well. thanks

Nothing yet. Hopefully someone could help us.

closed 2022-08-04 by song940