lsongdev/node-escpos #370: node-escpos prints twice

longGr, 2021-11-18:
I have the following code

`const escpos = require('escpos');
escpos.USB = require('escpos-usb');
const device  = new escpos.USB(2727, 772);
const options = { encoding: "ISO8859-15" /* default */ }
const printer = new escpos.Printer(device, options);

device.open(function(error){
  console.log("internal call")
  printer
  .text("Test123 This is a happy printer doing it's job twice. Too happy?")
  .close()
});
`

when I run the code with
`$node index.js`

the thermal printer will print the text twice. I have also tried to add the `cut()` `flush()` method within the open callback method. I can see that this callback method is called two times, as in the console it prints "internal call" twice. 

Expected: the printer only prints the text once. 

Thanks for the help

MLDMoritz, 2022-03-11:
Hey there, found a solution for the problem? Noticing the same behavior on a TH230.

longGr, 2022-03-11:
Kind of. Within the function that is define I the device.open() I checked for a flag which toggles each time the function is executed. So I it's some ugly workaround but at least until now it works just fine. 

So in m previous example, after the line console.log("internal call") I added the following:

`if(toggleSwitch){
    printer
    .text(...
}
toggleSwitch = !toggleSwitch`