lsongdev/node-escpos #376: TypeError: usb.on is not a function

firdausyah22, 2021-12-15:
Hello, i'v tried and use example code, and igot error like this

TypeError: usb.on is not a function.

My code :
const escpos = require("escpos");
escpos.USB = require('escpos-usb');

const device = new escpos.USB(1046, 20497);
// const device  = new escpos.RawBT();
// const device  = new escpos.Network('localhost');
// const device  = new escpos.Serial('/dev/usb/lp0');
const printer = new escpos.Printer(device);

console.log(device)

device.open(function (err) {
  printer
    .font("a")
    .align("ct")
    .style("bu")
    .size(1, 1)
    .text("The quick brown fox jumps over the lazy dog")
    .text("敏捷的棕色狐狸跳过懒狗")
    .barcode("1234567", "EAN8")
    .qrimage("https://github.com/song940/node-escpos", function (err) {
      this.cut();
      this.close();
    });
});

sam-lex, 2021-12-16:
this happens on windows 10
- on linux - ubuntu it's working fine.

if you're in a rush like me, a hacky-temporary workaround would be to go into the `node_modules/escpos-usb/index.js` and comment out `usb.on ...` part.

Then you'll run into another problem, saying `usb.removeAllListeners is not a function`. I commented that part out as well. Then it works ok-ish.

 I had already made an app with electron + express and stuff on Linux about a month ago. And just about 3 days ago I tested on a windows machine for the first time, and got these same errors...
I still got another problem to work with, which is the image I generate for it to print seems to "not fit" all of a sudden..

It seems to be somewhat related with this.. idk https://github.com/MadLittleMods/node-usb-detection/issues/77

Anyways.. good luck

romaia, 2021-12-16:
This actually seem to be related with a major release of `usb` [1]

From what I checked, this project didn't pin the dependency [2], and it was working fine with version 1.9.2

[1] https://www.npmjs.com/package/usb
[2] https://github.com/song940/node-escpos/blob/v3/packages/usb/package.json#L10

firdausyah22, 2021-12-17:
> this happens on windows 10
> 
> * on linux - ubuntu it's working fine.
> 
> if you're in a rush like me, a hacky-temporary workaround would be to go into the `node_modules/escpos-usb/index.js` and comment out `usb.on ...` part.
> 
> Then you'll run into another problem, saying `usb.removeAllListeners is not a function`. I commented that part out as well. Then it works ok-ish.
> 
> I had already made an app with electron + express and stuff on Linux about a month ago. And just about 3 days ago I tested on a windows machine for the first time, and got these same errors... I still got another problem to work with, which is the image I generate for it to print seems to "not fit" all of a sudden..
> 
> It seems to be somewhat related with this.. idk [MadLittleMods/node-usb-detection#77](https://github.com/MadLittleMods/node-usb-detection/issues/77)
> 
> Anyways.. good luck

oh wow, for now it's working. thanks

peteruithoven, 2022-02-09:
For npm using folks:
Even though npm 8 now supports [overrides](https://docs.npmjs.com/cli/v8/configuring-npm/package-json#overrides), that didn't really work for us, but installing usb as a direct dependency of our project worked. Using 1.9.2 like romaia mentioned.
```
npm install usb@1.9.2
```

flapili, 2022-05-04:
Same issue and using usb@1.9.2 did not help :/

AppPilots, 2022-05-11:
Same problem here... 
No solution at all.

flapili, 2022-05-13:
Look like usb@1.9.2 doesn't existe (anymore?)
https://www.npmjs.com/package/usb/v/1.8.8
Using 1.8.8 solved the issue

YovanggaAnandhika, 2022-05-18:
For Fix The Problem Please Migrate To USB Version 2
@see https://github.com/node-usb/node-usb#migrating-to-v200
@see https://github.com/song940/node-escpos/pull/388

i am change node_module/escpos-usb like 
`
let { usb } = require('usb');
`

Like my pull request

YovanggaAnandhika, 2022-05-18:
please close this issue

closed 2022-05-19 by song940

ferminc, 2022-08-16:
> go into the `node_modules/escpos-usb/index.js` and comment out `usb.on ...` part.

this worked for me, I also updated to usb 2.5.0, btw yes, this is still an issue

Siyavash-Adib, 2023-06-27:
> This actually seem to be related with a major release of `usb` [1]
> 
> From what I checked, this project didn't pin the dependency [2], and it was working fine with version 1.9.2
> 
> Using `yarn set resolution usb@npm:* 1.9.2 --save` for now fixed this for me
> 
> [1] https://www.npmjs.com/package/usb [2] https://github.com/song940/node-escpos/blob/v3/packages/usb/package.json#L10

Thank a lot, it's working for me now

ferminc, 2023-06-27:
I recommend you to use https://github.com/node-escpos/driver
The song940 node-escpos is basically deprecated by now

Using yarn set resolution ***@***.***:* 1.9.2 --save for now fixed this for me

Joshjoyb, 2024-09-15:
> this happens on windows 10
>
> * on linux - ubuntu it's working fine.
>
> if you're in a rush like me, a hacky-temporary workaround would be to go into the `node_modules/escpos-usb/index.js` and comment [...]

After doing this, I'm getting another error message that goes: "LIBUSB_ERROR_NOT_SUPPORTED" - what should I do?

sam-lex, 2024-09-16:
> After doing this, I'm getting another error message that goes: "LIBUSB_ERROR_NOT_SUPPORTED" - what should I do?

I would recommend you check out this one instead https://github.com/node-escpos/driver or some other solutions from other comments.

I myself am not using it anymore, so I have no idea. But regarding your error `LIBUSB_ERROR_NOT_SUPPORTED`, it's listed in here:
https://github.com/lsongdev/node-escpos/blob/v3/packages/usb/README.md

fuadriyadhi, 2025-08-14:
Why, when I comment usb.on, an error appears "LIBUSB_ERROR_NOT_SUPPORTED"