lsongdev/node-escpos #383: SerialPort new version

GuilhermeCouto, 2022-03-09:
Hi everybody, I found a problem with printing on Serial ports.

Serialprinter updates and so or file adapter escpos-serialprinter must be updated too.

The error you got is TypeError: SerialPort is not a constructor

Thats because the constructor changed.

I fixed it by changing the line 17 to `this.device = new SerialPort({ path: port, baudRate: 9600,autoOpen: false });`

If you could change the npm repo I will appreciate.

marceloch2, 2022-04-04:
Same here. The new version of SerialPort too.

ma-zal, 2022-04-10:
### Root cause

Library `escpos` has `package.json` dependencies defined without versions:

```json
{
  "optionalDependencies": {
    "serialport": "*",
  }
}
```

And `escpos` is designed for `serialport` version 9. And on 11 Dec 2021 there has been [released `serialport` version 10](https://github.com/serialport/node-serialport/releases), which has some breaking changes.

### Possible workaround

Force `serialport` library version 9.

1) Add to `package.json`:

    ```json
    {
      ... rest of file,
      "resolutions": {
        "escpos/serialport": "9.2.8"
      }
    }
    ```

2) Delete `node_modules` and `yarn.lock`.

3) Install all dependencies again by `yarn`.

closed 2022-08-04 by song940