lsongdev/node-escpos #385: UTF8 problem

AzeDonLees, 2022-04-03:
[image]
hello everyone, I tried to print a smiley face and Russian letters, nothing comes out. Does anyone have ideas?
i tried also 860, not have result.
Code:
const options = { encoding: "UTF8" };

const printer = new escpos.Printer(device, options);

device.open(function (error) {
  printer
    .size(0, 0)
    .font("a")
    .align("ct")
    .style("b")
    .text("привет андрей ❤️")
    .cut()
    .close();
});

AzeDonLees, 2022-04-03:
and ə ö ğ ı ə ç ş ü not work too

andrefmandrade, 2022-05-05:
use 860 instead of UTF8
`const options = { encoding: '860' };`

Louis-jk, 2022-09-16:
Would you like to try this?
```javascript
const options = { encoding: "cp866" };
```
Cyrillic encoding code : `cp866`