lsongdev/node-escpos #390: Can not find printer (vps) ubuntu 20.04

OussJouh, 2022-05-29:
Hey, 
i have a printer connected using usb and ethernet 
in localhost everything works good i can print using USB port (escpos-usb) also using ethernet (escpos-network)
after deploying my app to a vps ubuntu 20.04
my printer couldn't be found, USB.findPrinter() return empty array
knowing that i have installed sudo apt-get install build-essential libudev-dev as it is required
after checking connected usb devices in server using lsusb it doesn't return the connected usb printer

YovanggaAnandhika, 2022-08-27:
the module **escpos-usb** works if you run code to which your computer is connected locally.
You cannot perform a print command via usb if the device is not detected in the system manager on the specified device

if my guess is right, your connected device is on another device.
----------
as an alternative, you can use the **escpos-network** method by forwarding ip and some other things so that your printer can be accessed from outside the network using public ip and DDNS system

rizalbahriawan, 2023-10-09:
I also have this issue, I'm running locally from windows and using usb printer, and using zadig, and everything works fine. But when I deploy to server (using ubuntu), the printer isn't printing. Is this way (printing that deployed to server) only works by using network printer, and not usb printer?