lsongdev/node-escpos #403: How to setup this project?

dohooo, 2022-08-26:
I want to give some features, but I can't run build this project. Can you provide a detailed contribution? Thanks for your effort! Awesome project!

YovanggaAnandhika, 2022-08-27:
1. you can fork this repository for your costum repository name.
2. after repository fork created. create a new folder in your laptop for the costum project.
3. after create folder, if you use windows ``` cd /d <projectLocation> ``` and ``` cd <projectLocation>
4. run ```git clone <url repository>``` in terminal or command prompt.
5. modification code after clone successfully.
6. Publish Your code (@see)[https://docs.npmjs.com/creating-and-publishing-scoped-public-packages]

dohooo, 2022-08-27:
RE https://github.com/song940/node-escpos/issues/397#issuecomment-1229111060

YovanggaAnandhika, 2022-08-27:
[RE](https://github.com/song940/node-escpos/issues/397#issuecomment-1229249916)