lsongdev/node-escpos #406: Please ask whether node escpos supports parallel port LPT communication printing? Or node escpos does not support parallel port LPT. Is there an alternative?

wangshihuyue, 2022-09-15:
Hello, the current ongoing project uses Electron to develop desktop client. The project has printing requirements. Currently, the desktop client supports USB and serial port printing. The customer feedback needs to support parallel port LPT communication printing. Please ask whether node escpos supports parallel port LPT communication printing? If it is possible to give an example for reference, or node escpos does not support parallel port LPT, is there an alternative? Sorry to bother you.

khetkoong, 2022-10-05:
I got the same problem

Psychopoulet, 2022-10-06:
I had the same problem : on some old printers, there is some stupid rules which give the same VID/PID to the 2 interfaces of the printer, so the package is not able to find which one is for power supply / firmewire update and which one is for the actual escpos communication.

there is no de solution, because it's at the binary level of the usb package : libusb does not support that.
it give us the flag at the interface level which identify the 2, but it's unusable 'cause the call is on VID/PID

so, you have to install a software which emulate properly new 2 different VID/PID (my choice) for these interfaces, or by a more recent printer which has 2 different VID/PID

khetkoong, 2022-10-06:
I found this one, it works for me
https://github.774.gs/bigrocs/escpos-win/tree/815d3b8c5c2e9a37505ef74a748279edcf8acbac

pranavwani, 2024-12-13:
Hi @khetkoong Thanks man, It works.

pranavwani, 2024-12-13:
> I found this one, it works for me https://github.774.gs/bigrocs/escpos-win/tree/815d3b8c5c2e9a37505ef74a748279edcf8acbac

Hi @khetkoong Thanks man, It works.

Repo: https://github.com/bigrocs/escpos-win