lsongdev/node-escpos #409: Error in escpos-usb line 52 usb.on() not working

jmrg-link, 2022-10-04:
Error line **usb.on** comment line .
Printer not init .
escpos-usb -> index
line comments 
/*
usb.on('detach', function(device){
    if(device == self.device) {
      self.emit('detach'    , device);
      self.emit('disconnect', device);
      self.device = null;
    }
  });
*/

XavierTM, 2022-10-25:
I think the issue is being caused by the ```utils.inherit()``` which is deprecated. I fixed this by copying all the code in the ```escpos-usb``` module into a file, commenting out that part, then requiring that file instead.

apaleslimghost, 2022-12-31:
this is because `escpos-usb` has a wildcard dependency on `usb`, which means it's pulling in versions with breaking changes it's not compatible with. with recent versions of npm you can fix this by adding a version override for `usb` to your `package.json`:

```json
  "overrides": {
    "usb": "1.8.8"
  }
```

wxuefei, 2023-04-08:
you can try : usb.usb.on ...

denkol, 2023-10-05:
Was the same problem. @apaleslimghost solution **helps**.

Zetjen, 2024-11-11:
Fixed on #429

closed 2025-04-21 by jmrg-link