lsongdev/node-escpos #414: Can't fetch printer status for Usb device

saitahmety, 2022-10-27:
Using a usb printer adapter and trying to fetch the status of the printer.

```
"escpos": "^3.0.0-alpha.6",
"escpos-usb": "^3.0.0-alpha.4",
```

Here is my code creating the printer:
```
  const device = new escpos.USB(idVendor, idProduct);
  const options = {
    encoding: "cp857",
  };
  const printer = new escpos.Printer(device, options);
```

Here is the code:

```
  printer.getStatuses((status) => {
    console.log('status: ', status);
    console.log("status json: ", status.toJSON());
  });
```

This fails:
```
TypeError: this.adapter.read is not a function
     at Printer.getStatuses (/home/kiosk/kiosk-hub/node_modules/escpos/index.js:915:16)
     at /home/kiosk/kiosk-hub/routes/index.js:453:11
     at Layer.handle [as handle_request] (/home/kiosk/kiosk-hub/node_modules/express/lib/router/layer.js:95:5)
     at next (/home/kiosk/kiosk-hub/node_modules/express/lib/router/route.js:137:13)
     at Route.dispatch (/home/kiosk/kiosk-hub/node_modules/express/lib/router/route.js:112:3)
     at Layer.handle [as handle_request] (/home/kiosk/kiosk-hub/node_modules/express/lib/router/layer.js:95:5)
     at /home/kiosk/kiosk-hub/node_modules/express/lib/router/index.js:281:22
     at Function.process_params (/home/kiosk/kiosk-hub/node_modules/express/lib/router/index.js:335:12)
     at next (/home/kiosk/kiosk-hub/node_modules/express/lib/router/index.js:275:10)
     at Function.handle (/home/kiosk/kiosk-hub/node_modules/express/lib/router/index.js:174:3)
```

dohooo, 2022-10-27:
I have solved this problem. #412

saitahmety, 2022-10-31:
How can I use this new version in my project?

saitahmety, 2022-10-31:
> I have solved this problem. #412

How can I use this new version in my project?