lsongdev/node-escpos #415: node-escpos don't work with electron

Kaplas85, 2022-11-21:
Hi. I'm trying yo use this package with USB adapter and dosn't work. In development works well but builded (production) not

This is when I open the app 
[image]

my main.js code

```javascript
import { app, autoUpdater, dialog, ipcMain } from "electron";
import serve from "electron-serve";
import { createWindow } from "./helpers";
const escpos = require("escpos");
escpos.USB = require("escpos-usb");
const isProd = process.env.NODE_ENV === "production";

if (isProd) {
  serve({ directory: "app" });
} else {
  app.setPath("userData", `${app.getPath("userData")} (development)`);
}

(async () => {
  await app.whenReady();

  const mainWindow = createWindow("main", {
    width: 1000,
    height: 600,
  });

  if (isProd) {
    await mainWindow.loadURL("app://./home.html");
  } else {
    const port = process.argv[2];
    await mainWindow.loadURL(`http://localhost:${port}/home`);
    mainWindow.webContents.openDevTools();
  }
})();

ipcMain.handle("get-version", async (e) => {
  return app.getVersion();
});

ipcMain.handle("find-printers", async (e) => {
  const usbDevices = USB.findPrinter();
  console.log(usbDevices);
});

ipcMain.handle("print-receipt", async (e, itemsData, orderData) => {
  const device = new escpos.USB();
  const options = {
    encoding: "utf-8",
  };
  const printer = new escpos.Printer(device, options);

  device.open(async function (err) {
    if (err) {
      console.log(err);
      return;
    }

    printer.align("CT").size(10, 10).style("B").text("Dimalchi");
    printer
      .align("CT")
      .style("NORMAL")
      .size(1, 1)
      .text("Inv. Alma Fashion C.A.")
      .text("RIF: J40556539-0")
      .feed(1)
      .text(
        "Calle 100 (COLOMBIA) Casa Nro 101-10 Sector Centro de Valencia Valencia Carabobo Zona Postal 2001"
      );
    printer
      .feed(1)
      .align("LT")
      .style("B")
      .text(`Nota de Entrega: ${orderData.id}`)
      .style("NORMAL")
      .text(`Cliente: ${orderData.clientName}`)
      .text(
        `Fecha: ${new Date(orderData.createdAt).toLocaleDateString("es-ve")}`
      )
      .feed(2);

    printer
      .table(["Producto", "Cantidad", "Precio", "Total"])
      .drawLine()
      .feed(1);
    itemsData.map((item) => {
      printer.tableCustom([
        { text: item.name, align: "CENTER" },
        { text: item.quantity, align: "CENTER" },
        { text: item.price, align: "CENTER" },
        { text: item.total, align: "CENTER" },
      ]);
    });
    printer.feed(1).drawLine().feed(1);

    printer
      .text("Subtotal: Bs " + orderData.subTotal)
      .text("IVA: Bs " + orderData.ivaAmount)
      .text("Total: Bs " + orderData.total);

    printer.feed(2);
    printer.align("CT").text("¡Gracias por su compra!");

    printer.feed(4);
    printer.cut().close();
  });
});
```

YovanggaAnandhika, 2022-11-23:
i am same issue for latest version electron. no error and no catch

Kaplas85, 2022-11-24:
I'm using Electron v15 :(

dohooo, 2022-11-24:
I'll create an example for this later today with this [lib](https://github.com/node-escpos/driver).

Kaplas85, 2022-11-24:
Thanks @dohooo !

closed 2022-12-08 by Kaplas85