lsongdev/node-escpos #428: usb.on is not a function

JuanDa237, 2023-11-15:
Node v18.15.0

"dependencies": {
    "escpos": "^3.0.0-alpha.6",
    "escpos-usb": "^3.0.0-alpha.4"
 }
 
 Code:
 
 const escpos = require('escpos');
escpos.USB = require('escpos-usb');

const device = new escpos.USB(0x2BDF, 0x0280);
const printer = new escpos.Printer(device);

device.open(function (error) {
    if (error) {
        console.error('Error opening USB device:', error);
        return;
    }

    printer
        .font('a')
        .align('ct')
        .style('bu')
        .size(1, 1)
        .text('Your Store Name')
        .text('----------------')
        .align('lt')
        .text('Item 1       $10.00')
        .text('Item 2        $7.50')
        .text('----------------')
        .align('rt')
        .text('Total:      $17.50')
        .cut()
        .close();
});

Error:
C:\Users\juan\Desktop\test-printer\node_modules\escpos-usb\index.js:52
  usb.on('detach', function(device){
      ^

TypeError: usb.on is not a function
    at new USB (C:\Users\x\Desktop\test-printer\node_modules\escpos-usb\index.js:52:7)
    at Object.<anonymous> (C:\Users\x\Desktop\test-printer\index.js:5:16)
    at Module._compile (node:internal/modules/cjs/loader:1254:14)
    at Module._extensions..js (node:internal/modules/cjs/loader:1308:10)
    at Module.load (node:internal/modules/cjs/loader:1117:32)
    at Module._load (node:internal/modules/cjs/loader:958:12)
    at Function.executeUserEntryPoint [as runMain] (node:internal/modules/run_main:81:12)
    at node:internal/main/run_main_module:23:47

SZZH, 2023-11-15:
maybe you can try, usb module version to 1

SZZH, 2023-11-15:
> maybe you can try, usb module version to 1

pnpm i usb@1.x

giegiey, 2024-01-06:
npm install -S usb@1.8.0

joshuayoes, 2024-03-20:
You can also run apply this `patch-package` patch

Add this to `patch/escpos-usb+3.0.0-alpha.4.patch`

```
diff --git a/node_modules/escpos-usb/index.js b/node_modules/escpos-usb/index.js
index 8c21546..7b3aa42 100644
--- a/node_modules/escpos-usb/index.js
+++ b/node_modules/escpos-usb/index.js
@@ -25,7 +25,7 @@ const IFACE_CLASS = {
 function USB(vid, pid){
 
   if (!usb) {
-    usb = require('usb');
+    usb = require('usb').usb;
   }
 
   EventEmitter.call(this);
@@ -67,7 +67,7 @@ function USB(vid, pid){
  */
 USB.findPrinter = function(){
   if (!usb) {
-    usb = require('usb');
+    usb = require('usb').usb
   }
   return usb.getDeviceList().filter(function(device){
     try{
```

Then run `npx patch-package` in your project.

codewithshinde, 2024-04-04:
**This worked for me !!**

- go to node_modules/escpos-usb/index.js
- comment down the following lines

```
  // usb.on('detach', function(device){
  //   if(device == self.device) {
  //     self.emit('detach'    , device);
  //     self.emit('disconnect', device);
  //     self.device = null;
  //   }
  // });
```

Cu4rach4, 2024-06-14:
Hi, im fixed this using the latest version of usb.
    "usb": "^1.8.8"

then use Zadig driver.

Zetjen, 2024-11-11:
#429 This was merged and fixed

alexoxte, 2024-11-26:
> #429 This was merged and fixed

I use latest package for escpos-usb, but the error persist I try use directy github repository and branch on my package.json, but cannot use them because typescript code isn't transpiled. any suggestion for it?

Zetjen, 2024-11-26:
I see that the [npm registry](https://www.npmjs.com/package/escpos-usb) was updated 4 years ago and the same with [GH registry](https://github.com/lsongdev/node-escpos/releases), so first don't use the npm and GH package.

You can try to install from the repo directly. And then build it with `tsc`.

Zetjen, 2024-11-26:
I'll try to push a GH action to automatically deploy and compile the package.

alexoxte, 2024-11-26:
Maybe is necessary update version on package.json in ecspos-usb package

dann95, 2025-03-11:
is this project open to pr for update usage of node-usb to latest versions? on version 3 it has broken the api on the listener .on();
forcing users to set resolutions to node-usb max at 1.9.2, node-usb is 1.9.2 is a bit old and may cause some problem in complations like electron etc. If this project is open for pr, i can do it

clopezpro, 2025-04-02:
I use the version ^3.0.0-alpha.6 and "ESCPOS-USB": " ^3.0.0-alpha.4"
And I must even comment on the line

[image]
To work