lsongdev/node-escpos #437: [FEAT]: how to get printer list

Boolean2023, 2024-10-10:
### Description

is there a way to retrieve the list of printers paired on the machine with this package? if so, how do I get there?

### Additional Information

Printers list node-escpos

Psychopoulet, 2024-10-15:
with the embeded "usb" package, like this ?

usb.getDeviceList().filter((device) => { // has interfaces

    return device && device.configDescriptor && device.configDescriptor.interfaces && device.configDescriptor.interfaces.length;

}).map((device) => { // formate

    const interfaces = [];

    device.configDescriptor.interfaces.forEach((its) => {

        its.forEach((it) => {

            interfaces.push({
                "id": it.bInterfaceNumber,
                "type": it.bInterfaceClass
            });

        });

    });

    return {
        "bus": device.busNumber,
        "port": device.deviceAddress,
        "vid": device.deviceDescriptor.idVendor,
        "pid": device.deviceDescriptor.idProduct,
        "interfaces": interfaces
    };

}).filter((device) => { // only printers

    return !!device.interfaces.find((it) => {
        return 7 === it.type;
    });

});

tamil752002, 2024-10-23:
can i get Printer List , which is  Connect My Lan Network ?  is this Possible

Boolean2023, 2024-10-25:
OK Thanks

clopezpro, 2025-04-02:
estoy usando "escpos": "^3.0.0-alpha.6",
    "escpos-usb": "^3.0.0-alpha.4"

This works however in TypeScript this error apparently generates the USB interface awaits a string vine

[image]