lsongdev/node-escpos #441: [BUG]:  usb.on is not a function

lingyunzhizun, 2025-04-16:
### Description of the bug

```
\node_modules\escpos-usb\index.js:53
  usb.on('detach', function(device){
      ^

TypeError: usb.on is not a function
    at new USB (D:\GolandProjects\testprint\usb_printer_project\usb_printer_project_nodejs\node_modules\escpos-usb\index.js:53:7)
    at Object.<anonymous> (D:\GolandProjects\testprint\usb_printer_project\usb_printer_project_nodejs\index.js:5:16)    
    at Module._compile (node:internal/modules/cjs/loader:1565:14)
    at Object..js (node:internal/modules/cjs/loader:1708:10)
    at Module.load (node:internal/modules/cjs/loader:1318:32)
    at Function._load (node:internal/modules/cjs/loader:1128:12)
    at TracingChannel.traceSync (node:diagnostics_channel:322:14)
    at wrapModuleLoad (node:internal/modules/cjs/loader:219:24)
    at Function.executeUserEntryPoint [as runMain] (node:internal/modules/run_main:170:5)
    at node:internal/main/run_main_module:36:49

Node.js v22.12.0
```
```
 "escpos": "^3.0.0-alpha.6",
 "escpos-usb": "^3.0.0-alpha.4"
```

### Steps To Reproduce

直接使用示例
```
npm init -y
npm install escpos escpos-usb
node index.js
```

```
const escpos = require("escpos");
// install escpos-usb adapter module manually
escpos.USB = require("escpos-usb");
// Select the adapter based on your printer type
const device = new escpos.USB();
// const device  = new escpos.Network('localhost');
// const device  = new escpos.Serial('/dev/usb/lp0');

const options = { encoding: "GB18030" /* default */ };
// encoding is optional

const printer = new escpos.Printer(device, options);

device.open(function (error) {
    printer
        .font("a")
        .align("ct")
        .style("bu")
        .size(1, 1)
        .text("The quick brown fox jumps over the lazy dog")
        .text("敏捷的棕色狐狸跳过懒狗")
        .barcode("1234567", "EAN8")
        .table(["One", "Two", "Three"])
        .tableCustom(
            [
                { text: "Left", align: "LEFT", width: 0.33, style: "B" },
                { text: "Center", align: "CENTER", width: 0.33 },
                { text: "Right", align: "RIGHT", width: 0.33 },
            ],
            { encoding: "cp857", size: [1, 1] } // Optional
        )
        .qrimage("https://github.com/song940/node-escpos", function (err) {
            this.cut();
            this.close();
        });
});

```

AnSiwei, 2025-04-20:
去node_modules里把usb.on那一段注释掉就可以了

clopezpro, 2025-04-22:
You have to directly change a code line but this is not yet integrated into the official version check #428

clopezpro, 2025-04-22:
I have battled with this library, even in the types, but I manage to work thanks to the comments, Python's is another case

amodsachintha, 2025-11-16:
Actually, I was able to make this work by patching `escpos-usb` myself. Use `patch-package` to apply this fix.

```
    "escpos": "^3.0.0-alpha.6",
    "escpos-usb": "^3.0.0-alpha.4",
    "patch-package": "^8.0.1",
    "usb": "^2.16.0"
```

escpos-usb+3.0.0-alpha.4.patch
```
diff --git a/node_modules/escpos-usb/index.js b/node_modules/escpos-usb/index.js
index 8c21546..6343976 100644
--- a/node_modules/escpos-usb/index.js
+++ b/node_modules/escpos-usb/index.js
@@ -49,7 +49,7 @@ function USB(vid, pid){
   if (!this.device)
     throw new Error('Can not find printer');
 
-  usb.on('detach', function(device){
+  usb.usb.on('detach', function(device){
     if(device == self.device) {
       self.emit('detach'    , device);
       self.emit('disconnect', device);
@@ -166,7 +166,7 @@ USB.prototype.close = function(callback){
     try {
 
       this.device.close();
-      usb.removeAllListeners('detach');
+      usb.usb.removeAllListeners('detach');
 
       callback && callback(null);
       this.emit('close', this.device);

```